Philra94/homeassistant-hoymiles-cloud #1: not able to instal using HACS

Wind06, 2025-04-26:
Hello,

I am trying this integration, but get an error message using HACS " The version 3d4135b for this integration can not be used with HACS."

Thanks

Philra94, 2025-04-26:
Hi

closed 2025-04-26 by Philra94

Wind06, 2025-04-26:
Thanks for your quick answer, very much appreciated. In the mean time I installed it manually :-)

kluszczyn, 2025-05-03:
@Philra94 issue still exists, unable to install addon via HACS. Seems that you have to add `repository.yaml` at the root level - see https://developers.home-assistant.io/docs/add-ons/repository#repository-configuration

Krzysztonek, 2025-09-28:
> Seems that you have to add `repository.yaml` at the root level

**NOT true.** The `repository.yaml` file is **NOT applicable to HACS repositories**, but to **HA add-ons only**. 
If you want your repository to be accessible and installable from HACS, you need to create `hacs.json` file at the root level of this repository on your `Github.` account.