Philra94/homeassistant-hoymiles-cloud #6: API Error 3 (No Permission) when no battery is connected

Kaktusmensch, 2025-08-24:
​Hello,
I'm having issues setting up the integration. The debug log shows that the authentication is successful and the station is found.
However, the setup fails right after that when the integration tries to get battery settings. The API returns API error: 3 - No Permission.
​I do not have a battery connected to my Hoymiles system.
​It seems the integration does not handle the case where a user has no battery and fails because of the API error, instead of just skipping the battery part.
​I have attached the debug log. 

[error_log.txt](https://github.com/user-attachments/files/21955789/error_log.txt)

Thanks for your help!

Philra94, 2025-09-07:
@Claude please fix this and allow users with no battery to use the integration

Philra94, 2025-09-07:
@claude please fix this

commit 7900034b (2025-09-07): Fix API Error 3 (No Permission) for systems without battery

Handle the case where Hoymiles systems don't have a battery connected
by treating API error 3 - No Permission as an expected scenario. The
integration now logs this as info/debug instead of error and gracefully
falls back to default battery settings, allowing setup to complete
successfully.

Changes:
- hoymiles_api.py: Handle API error 3 as INFO instead of ERROR
- hoymiles_api.py: Change fallback warning to DEBUG with context
- __init__.py: Use DEBUG level for no-battery scenarios in coordinator

Fixes #6
closed 2025-09-07 by Philra94