Philra94/homeassistant-hoymiles-cloud #11: Hoymiles battery MS-A2 no battery power / flow direction / state of charge

humphrey4711, 2025-09-10:
Hello,
I have the hoymiles battery MS-A2 with the hoymiles app / cloud (role: diy user).
I get only the two fields "battery charge energy today" and "battery discharge enery today", all the other fields are 0 / empty:
battery power = 0
battery state of charge = unknown
battery flow direction = alway: discharging.
Can you help please ?

Philra94, 2025-09-13:
Hey,
Many thanks for your feedback!
Could you provide any logs to look into the issue? It is difficult to find the root cause that way.

humphrey4711, 2025-09-13:
The battery (MS-A2) ist the one and only device in the station.
All the "sensors" are reporting "no problem".
The "journal" reports only "Batterie Last Data Update Time" has changed.

In HAOS "system > protocols" I got this:

Logger: custom_components.hoymiles_cloud.sensor
Quelle: custom_components/hoymiles_cloud/sensor.py:427
Integration: Hoymiles Cloud (Dokumentation, Probleme)
Erstmals aufgetreten: 11:54:49 (632 Vorkommnisse)
Zuletzt protokolliert: 13:14:09

Error getting sensor value: invalid literal for int() with base 10: '12.0'
Error getting sensor value: could not convert string to float: '-'
Error getting sensor value: invalid literal for int() with base 10: '12.5'
Error getting sensor value: invalid literal for int() with base 10: '1.0'
Error getting sensor value: invalid literal for int() with base 10: '13.8'

Let me know  if you need more information.

humphrey4711, 2025-09-13:
great work the update works fine, no errors in system-protocol.
Now the sensors "state of charge", "today energy" and "total enery" are appearing.
But still no sensor for current power of the battery  like +/- watts, any idea ?

Philra94, 2025-09-14:
Can you see the current battery power on the hoymiles webpage or in the hoymiles app?

humphrey4711, 2025-09-25:
In the mqtt connection there are these two sensors:

[https://github.com/krikk/hoymiles-ms-a2-to-mqtt](url)

  - name: "Power from(+)/to(-) Battery"
    unique_id: hoymiles_msa2_power_to_from_battery
    state_topic: "homeassistant/sensor/MSA-2XXXXXXXXXXX/quick/state"
    value_template: "{{ value_json.bat_p }}"
    unit_of_measurement: "W"
    device_class: power
    state_class: measurement
    device:
      name: "MSA-2XXXXXXXXXXX"
      identifiers: "2XXXXXXXXXXX"
      manufacturer: "Hoymiles"
      model: "MS-A2"

  - name: "Battery State"
    unique_id: hoymiles_msa2_state
    state_topic: "homeassistant/sensor/MSA-2XXXXXXXXXXX/quick/state"
    value_template: "{{ value_json.bat_sts }}"
    device_class: enum
    options:
      - discharge
      - charge
      - standby
    device:
      name: "MSA-2XXXXXXXXXXX"
      identifiers: "2XXXXXXXXXXX"
      manufacturer: "Hoymiles"
      model: "MS-A2"

Philra94, 2025-10-07:
@claude please fix.

Summary

Symptoms: MS‑A2 shows only “Battery Charge/Discharge Energy Today”; “Battery Power” stays 0, SOC was unknown
(now fixed), and “Battery Flow Direction” is always discharging.
Cause: The integration expects battery real‑time fields in places that likely differ for MS‑A2. Specifically:
battery_power reads real_time_data.reflux_station_data.bms_power and is oddly gated on
is_battery_charging(...) being known, which can force it to 0.
battery_flow_direction relies on bms_power sign or a flows structure with a battery id of 4. MS‑A2 may
expose different keys and/or not provide flows in the same shape.
The earlier conversion errors (‘12.0’, ‘-’) align with the old parsing; your update addressed that with
safer converters.
Likely Data Shape Differences

MS‑A2 cloud payload appears to use keys like bat_p (W) and bat_sts (charge/discharge/standby) as seen in the
MQTT project you linked.
The current code looks for bms_power and infers direction from flows. If bms_power is absent and flows don’t
match expectations, direction falls back incorrectly and power reads 0.

Philra94, 2025-10-07:
@claude please look into this

Philra94, 2026-09-05:
Closing this — the remaining part of your report (battery power / flow direction never changing) is fixed on `main` by #49.

The API always returns `bms_power` as a positive magnitude and carries the charge/discharge direction separately in the `flows` array; the integration now reads that, so `battery_power` is signed (positive = discharging, negative = charging) and `battery_flow_direction` tracks the real state instead of being stuck on one value.

That matches the `Power from(+)/to(-) Battery` behaviour you referenced from the MQTT bridge. Please reopen if your MS-A2 still shows a stuck value after updating.

closed 2026-09-05 by Philra94